stefanor/signalk-victron-ble #4: Add support for Solar Power on MPPT

steveuk71, 2023-03-30:
Following fields available to report, the current Solar Input Watts and Total Yield Watts today    

"model_name": "SmartSolar MPPT 75|15",

    "solar_power": 0,

    "yield_today": 20

stefanor, 2023-04-05:
I'm afraid there aren't defined keys for either of these in the SignalK spec: https://signalk.org/specification/1.7.0/doc/vesselsBranch.html

The first step would be to define them in https://github.com/SignalK/specification

stefanor, 2023-04-05:
You can, of course, calculate both of these from existing values. Power = Voltage * Current, and yield today is Power since midnight.

stefanor, 2023-04-14:
`panelPower` is added in #6

stefanor, 2023-04-14:
Filed SignalK/specification#653 for the specification updates.

tkurki, 2023-04-15:
Just to make sure you know: the SK server is schema agnostic, you can basically invent your own paths. Naturally if the name of the path is obvious / noncontroversial we should pretty quickly add it to the specifiation. The main advantags are that then you get unit information from the schema and various pieces of software use the same path for the same thing.

But you *can* start using any path and SK server will work.

PR #6 (2023-04-15): extend SolarCharger output
although panelPower is non-spec (panelVoltage and panelCurrent are...), it's really helpful to see at a glance how well the panel is performing.

loadCurrent is in spec: https://signalk.org/specification/1.7.0/doc/vesselsBranch.html#vesselsregexpelectricalsolarregexploadcurrent 
and there's no reason not to populate the path if we have the data.

Fixes: #4
closed 2023-04-15 by stefanor