stefanor/signalk-victron-ble #8: Victron Smart Battery Sense - not reconised

raytechnical, 2023-09-17:
I am running version 0.0.5 of the application on a Raspberry Pi 4. The Smart Solar and BMV712 is working well with the app, I have since added a Smart Battery Sense to the system but it is not showing up in the system. 

I have had a look on Bluetoothctl on the Pi  and can see the Smart Sense appearing on the screen with the rest of the devices on the system. 

I have checked the SignalK logs and there is no errors or any mention of the Smart battery sense device.

Any pointers would be useful as to why the device is not being seen by the application?

Many thanks.

raytechnical, 2023-09-17:
I have since run victron_ble  on the Pi and I was able to read the data direct from the Battery Smart Sense sensor if that helps?

kmon6369, 2023-11-29:
The DC to DC is working great.  I have not been able to get the data from the Battery Sense.   Have tried the MPPT controllers but that is not working and understand that it has not been tested.

LasseFrisk, 2023-11-30:
With the change in my previous post the smart battery sense is now working.
About the MPPT controllers, my smart solar 100/20 is working.

kmon6369, 2023-11-30:
I will try the battery sense again and see if maybe I need to wait longer for it to connect.  The DC to DC can take up to 15 minutes for the plugin to see it.  I will also try again with the MPPT controllers.  I have 5.  3 are 100/30 and two are 100/50.

stefanor, 2023-12-08:
Sorry, I missed a line of code. Bought a battery sense and immediately spotted the problem :)

commit 7c7ef2b4 (2023-12-08): Hook up Battery Sense

Closes: #8
closed 2023-12-08 by stefanor