stefanor/signalk-victron-ble #9: MPPT values have the wrong units

elafargue, 2023-11-19:
Using a Victron MPPT, I noticed that all the values that are logged to SignalK are in volts, including current, yield, panel power, charging mode, etc...

stefanor, 2023-11-22:
Values are logged numerically. The units are specified by the signalk spec

elafargue, 2023-11-22:
Oh right, so it looks like the SignalK spec doesn't know the right units for those measurement types, then? Happy to open a PR there if you point me in the right direction...

paalkr, 2024-06-18:
I noticed that all values are now registered with the correct unit, except for the yieldToday. This shows up in SignalK as J (joule) instead of Wh. The actual value is identical to what the Victron app displays, but in the Victron app the unit is actually Wh.

paalkr, 2024-07-10:
I notice this is fixed in the code, but the fix is not included with the most recent release. Would it be possible to cut a new release with all the most recent changes to the code?