stefanor/signalk-victron-ble #12: SmartLithium support

orymate, 2024-03-24:
keshavdv/victron-ble#47 adds support for SmartLithium batteries, which would be nice to support with SignalK.

Unfortunately I can't use this plugin because I couldn't make the python lib working on my cerbo/VenusOS. So I run a different wrapper around the python library, and feed SignalK through deltas over TCP. The mapping I do may be useful however:
https://github.com/orymate/smart-lithium-ble-sk/blob/main/smart_lithium_ble_sk/scanner.py#L35-L50

gregsyoung, 2024-05-26:
i note that keshadev has added support for smart batteries,
also saw a msg from @j9brown that he was adding and merging?
.. but trying to understand the current status of support in signalk-victron-BLE plugin?

ive tested but it doesnt appear to find my smart batteries

stefanor, 2024-05-26:
> I note that keshadev has added support for smart batteries,

Where? https://github.com/keshavdv/victron-ble/pull/47 hasn't landed yet

gregsyoung, 2024-05-26:
did i misread this?

https://github.com/keshavdv/victron-ble/pull/49

stefanor, 2024-05-26:
That hasn't landed yet, either.

gregsyoung, 2024-05-26:
ah… ok .. im not overly familiar with github, 
push, pull , metge etc requests/status

… my bad.

but definitely looking forward to test with my smart batteries :0

gregsyoung, 2024-05-26:
and yes, i have two distributors in my system that will test its functionality also

commit fdeee677 (2024-10-24): Support Smart Lithium Batteries

Unfortunately SignalK doesn't define any schema for individual cell
voltages, so the most useful thing we can get from the batteries over
bluetooth isn't used.

Fixes: #12
closed 2024-10-24 by stefanor