stefanor/signalk-victron-ble #13: Unstable and freezes

paalkr, 2024-04-10:
I'm using the plugin with SignalK 2.7.0 on a Raspberry Pi 3B+. The plugin can run for several hours, and then just stops reporting values without any error messages in the logs. Everything else in SignalK seems to be running just fine, so it's only the Victron BL plugin that stops.

The last log entry just looks totally normal.

```
Apr 10 18:21:20 signalk signalk-server[9574]: 2024-04-10T16:21:20.888Z signalk-victron-ble {"updates": [{"source": {"label": "Victron", "type": "Bluetooth", "src": "XX:XX:XX:XX:XX:XX"}, "timestamp": "2024-04-10T16:21:20.887456Z", "values": [{"path": "electrical.solar.mppt-100-50.voltage", "value": 12.52}, {"path": "electrical.solar.mppt-100-50.current", "value": 0.0}, {"path": "electrical.solar.mppt-100-50.chargingMode", "value": "off"}, {"path": "electrical.solar.mppt-100-50.panelPower", "value": 0}, {"path": "electrical.solar.mppt-100-50.loadCurrent", "value": null}, {"path": "electrical.solar.mppt-100-50.yieldToday", "value": 0}]}]}
```

Restarting SignalK brings the plugin back to life, until it stops again after approx 10 hours.

What can be done to make the plugin run stable over time?

stefanor, 2024-04-10:
I'm afraid I haven't seen this behaviour, so you're probably going to have to debug it for me.

Can you still see it running in `ps`? Can you see what it got stuck on, with `strace` or `gdb`?

stefanor, 2024-04-10:
You could also try running it standalone, with the `test` harness. You can create a `test-devices.json` file, with contents like:

```
{
  "devices": [
    {
      "id": "house",
      "mac": "D9:8F:D1:D8:CB:F2",
      "key": "9ee0db100da5ce487156ac665c399ddb",
      "secondary_battery": "start"
    },
   { ... }
  ]
}
```

Listing all your devices that you have registered in SignalK. Then run `./test`.

paalkr, 2024-04-12:
Thanks for the feedback. I think the problem was on my side. After making a few adjustments on the Bluetooth stack in my RPi and changing the Load Output Operation Mode in my SmartSolar to `Always on`, the plugin has now run without interruption for several days. I think the changing Load Operation Mode in fact was the important part.

Thanks for a great plugin and quick response!

closed 2024-04-12 by paalkr