stefanor/signalk-victron-ble #16: TypeAlias ImportError: cannot import name 'TypeAlias' from 'typing'

peter-bothe, 2024-10-27:
Hi,

I am running the plugin on SignalK 2.11.0 on an Openplotter 3 on a Rasperry Pi 4.

After installing plugin version 0.1.0 there is this error, when I restart SignalK:

```
Oct 25 17:34:47 Traceback (most recent call last): File "/home/pi/.signalk/node_modules/signalk-victron-ble/plugin.py", line 8, in <module>  
Oct 25 17:34:47 from typing import Any, Callable, TypeAlias ImportError: cannot import name 'TypeAlias' from 'typing' (/usr/lib/python3.9/typing.py)  
Oct 25 17:34:47 Plugin exited 1, restarting...
```

I have installed Python 3.10 as an additional version and set it as the alternative for python3. I removed the plugin from SignalK and reinstalled it to setup the ve correctly. Now the plugin runs, but all openplotter (v3) apps fail to run, because they still need python3 to be V3.9.

So currently I have to configure Python 3.9 for python3, when I want to run an openplotter app and Python 3.10 to restart SignalK.

I guess that's not the intended way of use... 😉 

- What am I doing wrong?
- What can I do to solve the problem (except reverting to the previous version and missing all the good stuff - like seeing my Phoenix inverter).

Peter

stefanor, 2024-10-27:
Thanks for the report.

I made an assumption that we wouldn't be using Python 3.9 (as I'd assumed people were using current RPi OS versions). But if that's a common use-case, let's try to support it, it's not hard.

closed 2024-10-27 by stefanor

peter-bothe, 2024-10-27:
Thanks Stefano, works like a charm!

commit e6e69bac (2024-10-27): Restore support for Python 3.9

Fixes: #16