stefanor/signalk-victron-ble #24: Python 3.9 Compatibility Issue – | Type Hint Not Supported

Leijoma, 2025-05-15:
The below text is generated by chat GPT because i could not solve the issue by my self. That means the whole thing could be wrong, but i got it working this way.

Summary:
The plugin uses modern Python 3.10+ type hinting syntax with the union operator (|), which causes a crash on systems running Python 3.9 or earlier — a common setup for Signal K users on Raspberry Pi.

Error Details
Error message:

`TypeError: unsupported operand type(s) for |: 'type' and 'type'`

Relevant line (plugin.py):

`data: dict[str, str | float | None]`

Cause
The str | float | None syntax for type unions was introduced in PEP 604 and is only valid in Python 3.10 and later.

Older Python versions (3.9 and earlier) expect the traditional Union[...] syntax from the typing module.

Suggested Fix
Replace this:

`data: dict[str, str | float | None]`

With this:

```
from typing import Union

data: dict[str, Union[str, float, None]]
```

Why This Matters
Many Signal K installations — especially on Raspberry Pi — still run Python 3.9. Ensuring compatibility improves adoption and reduces friction for users setting up the plugin without upgrading their base OS.

stefanor, 2025-05-16:
Sorry, I realized just after releasing that this would be an issue.

closed 2025-05-16 by stefanor

Leijoma, 2025-05-16:
Thanks! /Magnus