stefanor/signalk-victron-ble #31: Data is in Wh, we shouldn't be applying coulomb() here.

kaospctqc, 2025-08-21:
https://github.com/stefanor/signalk-victron-ble/blob/d8198351f97deba86718cdad2e0b048ab20cd76c/plugin.py#L430

Hi, data.get_yield_today() is already in Wh, it is wrong to apply coulomb() to it.

stefanor, 2025-08-21:
This is actually reporting Joules, not Coulombs. The conversion is the same (*3600), because we're changing the time unit from Seconds to Hours.

stefanor, 2025-08-21:
Clarified in 5871e40

closed 2025-08-21 by stefanor

kaospctqc, 2025-08-21:
If you check this line, the comment says it's watt hours.
https://github.com/keshavdv/victron-ble/blob/f9a29f0d0e02ba2cbc2bdee4cd3058c7416a2c87/victron_ble/devices/solar_charger.py#L39

stefanor, 2025-08-21:
Yes. And we convert to Watt seconds (Juoles)

kaospctqc, 2025-08-21:
Thank you for the quick reply, sorry I forgot to say thanks initially :)

kaospctqc, 2025-08-21:
Ah, apologies, I misunderstood and thought you wanted to report Watt hours :) 
Thank you again :)

stefanor, 2025-08-21:
This isn't a defined signalk metric, but all the other metrics use seconds-based units: https://signalk.org/specification/1.7.0/doc/vesselsBranch.html